RetroPie/RetroPie-Setup #3338: Incorrect Resolution Set in /dev/shm/retroarch.cfg

theofficialgman, 2021-05-12:
Screen is rotated with xrandr left at boot time on this system (nintendo switch)

Retropie incorrectly populates the /dev/shm/retroarch.cfg with
```video_refresh_rate = "60"
video_fullscreen = "true"
video_fullscreen_x = "720"
video_fullscreen_y = "1280"
```
the X and Y values are backwards and causes all retroarch emulators to crash with output such as 
```/opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-nestopia/nestopia_libretro.so --config /opt/retropie/configs/nes/retroarch.cfg "/home/garrett/RetroPie/roms/nes/Super Mario Bros. (W) [!].nes" --appendconfig /dev/shm/retroarch.cfg
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  152 (XFree86-VidModeExtension)
  Minor opcode of failed request:  10 (XF86VidModeSwitchToMode)
  Value in failed request:  0x193
  Serial number of failed request:  31
  Current serial number in output stream:  33
  ```
  reversing the X and Y values manually in this file and launching with the same command works fine
  this is not an issue for display configurations where the screen is not rotated
  
  https://github.com/RetroPie/RetroPie-Setup/blob/d6d715b8867f101d2ef6cdf66dec23f20d0aa2ad/scriptmodules/supplementary/runcommand/runcommand.sh#L1089
  https://github.com/RetroPie/RetroPie-Setup/blob/d6d715b8867f101d2ef6cdf66dec23f20d0aa2ad/scriptmodules/supplementary/runcommand/runcommand.sh#L944
  applicable code linked above

theofficialgman, 2021-05-12:
output from xrandr --verbose to help debug the issue
```xrandr --verbose
Screen 0: minimum 8 x 8, current 1280 x 720, maximum 16384 x 16384
DSI-0 connected primary 1280x720+0+0 (0x18d) left (normal left inverted right x axis y axis) 0mm x 0mm
	Identifier: 0x18c
	Timestamp:  2192869
	Subpixel:   unknown
	Gamma:      1.0:1.0:1.0
	Brightness: 1.0
	Clones:    
	CRTC:       0
	CRTCs:      0
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
	_MUTTER_PRESENTATION_OUTPUT: 0 
	TegraOverlayBlendmode: Opaque 
		supported: Opaque, SourceAlphaBlend, PremultSourceAlphaBlend
	TegraOverlayPriority: 255 
		range: (0, 255)
	BorderDimensions: 4 
		supported: 4
	Border: 0 0 0 0 
		range: (0, 65535)
	SignalFormat: TMDS 
		supported: TMDS
	ConnectorType: Panel 
	non-desktop: 0 
		supported: 0, 1
  720x1280 (0x18d) 77.905MHz +HSync +VSync *current +preferred
        h: width   720 start  856 end  928 total 1000 skew    0 clock  77.91KHz
        v: height 1280 start 1290 end 1291 total 1300           clock  59.93Hz
DP-0 disconnected (normal left inverted right x axis y axis)
	Identifier: 0x18e
	Timestamp:  2192869
	Subpixel:   unknown
	Clones:    
	CRTCs:      1
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
	_MUTTER_PRESENTATION_OUTPUT: 0 
	TegraOverlayBlendmode: Opaque 
		supported: Opaque, SourceAlphaBlend, PremultSourceAlphaBlend
	TegraOverlayPriority: 255 
		range: (0, 255)
	BorderDimensions: 4 
		supported: 4
	Border: 0 0 0 0 
		range: (0, 65535)
	SignalFormat: DisplayPort 
		supported: DisplayPort
	ConnectorType: DisplayPort 
	non-desktop: 0 
		supported: 0, 1
```

joolswills, 2021-06-27:
This code was done by @psyke83 - maybe he has an idea

The code grabs the primary display and all modes supported by it then references them via the connected display ID (0x18d in this case).

Although the first reference has 1280x720 the first mode is actually 720x1280.

We might be able to workaround it by adding an option to ignore xrandr completely or swap the values based on the string "inverted" , but not sure the best fix if retroarch requires the values swapped as it would seem correct how we have them set in regards to the orientation.

joolswills, 2021-06-27:
I will need to look at the retroarch code also.

psyke83, 2021-06-27:
I think that this part of the log signifies the orientation:

"DSI-0 connected primary 1280x720+0+0 (0x18d) **left** (normal left inverted right x axis y axis) 0mm x 0mm"

The possible orientation types reported by xrandr are normal, left, right and inverted. When I have time, I'll check the retroarch config to see if there's an additional setting that we need to map to handle the specific orientation type, as simply doing an x/y resolution flip may not work for everybody.

theofficialgman, 2021-06-29:
If I might ask, what is the reason for this functionality at all?
setting the resolution values to empty eg: video_fullscreen_x="" seems to work fine at least for my config

theofficialgman, 2022-07-21:
@psyke83 can you look at this again? still an issue

theofficialgman, 2022-07-21:
basically the issue is this function https://github.com/RetroPie/RetroPie-Setup/blob/f271ce9dee01fb224aaac44d6e9c1230d1ab5087/scriptmodules/supplementary/runcommand/runcommand.sh#L272-L323

doesn't take into account rotation values left/right/inverted etc, AT ALL
all it does is report back the actual screen resolution (which doesn't factor in rotation)