RetroPie/RetroPie-Setup #3379: xpad triggers_to_buttons=1 patch breaks SDL gamepad mappings

cgutman, 2021-07-26:
Hello, I am the upstream maintainer of [Moonlight Embedded](https://github.com/moonlight-stream/moonlight-embedded). I am working on tracking down a user-reported issue with Xbox 360/One gamepads being mapped incorrectly while running inside RetroPie (and working perfectly fine when launched outside of RetroPie).

I tracked down the cause to the custom xpad which unconditionally maps triggers to button rather than analog axes:
https://github.com/RetroPie/RetroPie-Setup/blob/6d396a3d5b70417869eddcb6d3d74324d5fce689/scriptmodules/supplementary/xpad.sh#L49-L51

https://github.com/RetroPie/RetroPie-Setup/blob/6d396a3d5b70417869eddcb6d3d74324d5fce689/scriptmodules/supplementary/xpad/01_enable_leds_and_trigmapping.diff#L12-L27

The `triggers_to_buttons` option was never meant to be used in this way, exactly because it can break unaware userspace applications. Upstream xpad only allows it to be used for unknown gamepads (and even for that it is not recommended, because many applications and libraries like SDL have heuristic mappings for things resembling Xbox controllers that `triggers_to_buttons=1` will break).

This screws up any SDL gamepad mappings used by applications (like Moonlight) running inside RetroPie, because the SDL gamepad mapping GUID is identical to the version where triggers are mapped as axes. As a result of the change in number of axes and buttons, the mapping becomes completely wrong (not just the triggers).

Additionally, it prevents the use of the triggers as analog inputs for things like racing games where that functionality is important.

Analog triggers are the norm for all modern gamepads (Xbox 360/One/Series X, [DualShock 4](https://github.com/torvalds/linux/blob/master/drivers/hid/hid-sony.c#L424-L425), [DualSense](https://github.com/torvalds/linux/blob/ff1176468d368232b684f75e82563369208bc371/drivers/hid/hid-playstation.c#L926-L927)), and hid-sony/hid-playstation don't allow you to opt out of it.

Can we nuke this `triggers_to_buttons=1` change?

cmitu, 2021-08-03:
Thank you for the comprehensive write-up and the debugging done to diagnose the problem.

Looking at the history of the RetroPie's included `xpad` driver, looks like the reason for the patch was EmulationStation (the gaming front-end used by RetroPie) didn't properly support the trigger axis on the Xbox gamepads ([1]).

In order for the patch to be dropped - or just not install the modified `xpad` driver it by default - we'll have to make sure first that EmulationStation will support the trigger axis properly on Xbox (`xpad` supported) gamepads.

Note that the user can always uninstall the driver - the system will revert to the in-kernel `xpad` driver. Further updates will not re-install the driver automatically. 

Removing the patch altogether without proper support would probably break upgrading users, which have a configuration based on the previous `xpad` driver behavior, so most likely once we have proper support in EmulationStation, we can just skip installing the modified driver on new installations and update our docs.

[1] `xpad` patch - https://github.com/RetroPie/RetroPie-Setup/commit/ab7871d155c8fe393b50dbc58ef10de65bf20e5a#diff-d3abe0095936112ef6abe7c5dce7d57b236a7a1222f877b9539cca0b5468f03a

joolswills, 2021-08-03:
Sorry I didn't get back on this issue sooner but @cmitu covered everything. Thanks also for the detailed report.

cgutman, 2021-08-03:
> Thank you for the comprehensive write-up and the debugging done to diagnose the problem.
>
> Looking at the history of the RetroPie's included `xpad` driver, looks like the reason for the patch was [...]

Yep, I did check that and it does work today. I believe it was fixed back in 2019 by https://github.com/RetroPie/EmulationStation/commit/93fdfaa9c2815edd34ed79883d5a983eefde17d1

This image was captured while mapping my Xbox One S controller without the modified `xpad` driver. The analog trigger axes from the stock `xpad` driver can be mapped successfully in EmulationStation:
[image]

> Removing the patch altogether without proper support would probably break upgrading users, which have a configuration based on the previous `xpad` driver behavior, so most likely once we have proper [...]

Yeah I noticed that all the button mappings for EmulationStation were wrong after going back to the stock `xpad`. It made it tricky to get back into the input mapping menu to be able to fix it.

Without some special upgrade handling for old mappings, it definitely will break users if the default is changed in an update. It could probably be done by deleting any gamepad mappings where the name is in [the list in xpad](https://github.com/torvalds/linux/blob/d5ad8ec3cfb56a017de6a784835666475b4be349/drivers/input/joystick/xpad.c#L115-L339), but it certainly wouldn't be pretty.

Unless there are any better ideas, skipping the modified driver seems like a good compromise. It will at least stop new users from being affected.

cmitu, 2021-08-06:
> This image was captured while mapping my Xbox One S controller without the modified `xpad` driver. The analog trigger axes from the stock `xpad` driver can be mapped successfully in EmulationStation:

Thanks for testing - it looks promising. I'll have to find an Xbox controller to re-test this and check the input mapping scripts that RetroPie uses to configure RetroArch and a few other emulators based on the EmulationStation's config. If the code is there, then it will be easier to implement a resolution.

I'm more worried about clones and various 'smart' controllers (8Bitdo) with have multiple connection modes which may not function correctly, I know that we received a few complaints in the forums when the RetroPie `xpad` driver was not installed and the users complained about not being able to map the trigger axis.

HerbFargus, 2021-08-06:
Historically there were trigger issues with some Logitech controllers as well which may also be impacted

cgutman, 2021-08-06:
The logic introduced for mapping axis Z and axis RZ (2 and 5) in https://github.com/RetroPie/EmulationStation/commit/93fdfaa9c2815edd34ed79883d5a983eefde17d1 is controller-agnostic. It's not looking at the controller name or axis count, so third party `xpad` gamepads should not cause problems (even those with special mapping flags like `MAP_DPAD_TO_BUTTONS` or `MAP_TRIGGERS_TO_BUTTONS`).

`xpad` exposes the set of axes and buttons for all gamepads of each Xbox generation that it supports (subject to a few exceptions like fightsticks and dancepads with special `MAP_*` options in [xpad's list](https://github.com/torvalds/linux/blob/d5ad8ec3cfb56a017de6a784835666475b4be349/drivers/input/joystick/xpad.c#L115-L339)).

cmitu, 2021-08-10:
So I tested an old Xbox One controller and @HerbFargus tested his old Xbox360 & friend controllers and we can confirm that EmulationStation behaves correctly and detects/maps the Z/RZ axis. Clearly things have improved since the `xpad` patch that forced the `triggers_to_buttons` module parameter as a workaround.

 Now, I don't have other Xbox controllers, but I've looked at other similar/compatible controllers in the RetroArch's [udev](https://github.com/libretro/retroarch-joypad-autoconfig/tree/master/udev) configuration profiles and they are quite a few of them (42 if I counted correctly). I could probably get a similar list from SDL's gamescontrollerdb, but I'm not so familiar with the syntax.

   Some of the clones/fightsticks that are Xbox compatible don't correctly expose the full axis range as a genuine controller does (i.e. arcade sticks like _MadCatz FightStick TE_ or _Mad Catz BrawlStick_). Some of them are handled by `xpad`, as referenced by @cgutman [here](https://github.com/torvalds/linux/blob/d5ad8ec3cfb56a017de6a784835666475b4be349/drivers/input/joystick/xpad.c#L115-L339), but exceptions can show up with clones or fightsticks (like the recently released 8BitDo arcade stick reported [here](https://retropie.org.uk/forum/topic/28595/) in RetroPie's forums).

 So, as far as EmulationStation is concerned, we could forego without installing the patched `xpad` by default. 

As a deprecation plan, we could:
 - release the next RetroPie version (image) based on the current Raspberry Pi OS image (Debian Buster) with `xpad` installed as default, preserving the current behavior. Document the issues that the patched `xpad` has with Moonlight (and possibly PPSSPP/Steamlink or other emulator we offer which uses SDL's game controller API).
 -  use the Raspberry Pi OS next version, based on Debian Bullseye (which should be _soon_), to release the subsequent image, without RetroPie's patched `xpad` driver installed as default and make the change clear in the release notes. Document the new (correct) behavior and suggest either the patched driver or contacting the forums in case of issues. 
 -  add some verbiage/information/warning during the driver installation/uninstallation to suggest a re-configuration of the controller in EmulationStation.

closed 2024-03-23 by joolswills