RetroPie/RetroPie-Setup #3411: Update SDL1 script for Debian 11

alessandrofrancesconi, 2021-10-05:
Debian 11 Bullseye has been released as the new stable.
The function to retrieve the correct SDL1 package ([get_pkg_ver_sdl1](https://github.com/RetroPie/RetroPie-Setup/blob/7efdc0e43c5550feeada9f0f34d7eadf314436de/scriptmodules/supplementary/sdl1.sh#L18)) should handle the new Debian version "11".

closed 2021-10-07 by joolswills

kraoc, 2021-11-18:
A workaround I used is detailed here:
https://forums.raspberrypi.com/viewtopic.php?p=1940160#p1940160

alessandrofrancesconi, 2021-11-18:
> A workaround I used is detailed here: https://forums.raspberrypi.com/viewtopic.php?p=1940160#p1940160

That's cool... I'm the author of that forum thread, and I hadn't seen that post yet :)
I will try

kraoc, 2021-11-18:
RetroPie up & running on my Pi 4B :)

Still some issues compiling some tools... as of Retropie Manager which depends on python-dev and I didn't found a workaround.
But this tool seems deprecated so not really a problem.