RetroPie/RetroPie-Setup #3428: Seeking info on rpiOS changes and how they will effect retropie

Codythedragonrude, 2021-11-18:
I did not know where else to ask so here it is 

With rpiOS changing their base to debian 11 and adding the mutter compositor will any of these changes be coming to retropie for pi 4/400? 

The more important one for me would be mutter since whatever compositor or lack of compositor retropie currently uses is pretty bad at eliminating screen tearing 

This has been a roadblock for me since the pi3 honestly 

I like retropie but the screen tearing just makes it impossible for me to use 

So yeah 
just wondering if you are looking into adopting mutter compositing for the pi 4 2gb and up ram models?

cmitu, 2021-11-18:
> I did not know where else to ask so here it is

As the issue template says - use the forum for support questions or issues, the issue tracker is for actual bugs.

> just wondering if you are looking into adopting mutter compositing for the pi 4 2gb and up ram models?

RetroPie doesn't use the desktop environment and it's not included in the base image. You can however install it as a Port - see https://retropie.org.uk/docs/Adventure-Game-Studio/#install-pixel-desktop-environment - if you wish. This will probably still work when RetroPie will use the new Raspberry Pi OS release - so whatever RPI OS ships as desktop environment (mutter/openpbox) will be available as a port.

closed 2021-11-18 by cmitu

Codythedragonrude, 2021-11-18:
Do you have any pointers about screen tearing? 
Maybe info about the KMS Driver and stuff?