RetroPie/RetroPie-Setup #3468: Duckstation Completely gone

Redcarred2, 2022-02-04:
Duckstation got archived on GH, and the download link is broken for the libretro core, causing the setup script to error.

cmitu, 2022-02-04:
Yes, we're aware of that, it was already reported in the RetroPie forum.

Codythedragonrude, 2022-02-07:
> Duckstation got archived on GH, and the download link is broken for the libretro core, causing the setup script to error.

Duckstation is done the dev has put the github repo into read only mode and libretro have since stated that the only way the project would continue is in the form of swanstation 

so rather than worry about implementing duckstation i would wager they should focus on swanstation instead

no point in supporting a project were the creator locks the repo down anytime he gets mad about something

dankcushions, 2022-02-07:
this is not a discussion forum, and regardless keep this drama out of here please. we are aware of the issue.

cobalt2727, 2022-02-08:
I have binaries of what I'm pretty sure are the latest versions for amd64 and aarch64 on my devices

@stenzek is there any issue with me uploading them to share with the RetroPie devs, or would you prefer RetroPie users just used other PSX emulators given everything that's happened?

Codythedragonrude, 2022-02-08:
> this is not a discussion forum, and regardless keep this drama out of here please. we are aware of the issue.

you honestly can't be expecting everyone to make a forum account when everyone has moved onto discord and github to talk about issues

i like retropie and the work you guys have done but the ways you give us to even respond to stuff like this is archaic and bad

cmitu, 2022-02-11:
Should be fixed by https://github.com/RetroPie/RetroPie-Setup/commit/495ac8d.

closed 2022-02-11 by cmitu