RetroPie/RetroPie-Setup #3482: Fix for SDL2 compile in Raspberry Pi OS 64 bit Bullseye

Luciano2018, 2022-02-15:
I was banned by cmitu so i cant post in forum sorry, just for share this fix i found that make posible compile sucessfully SDL2 in the mentioned system that is unsupported but works.
I follow the normal steps for building from retropie page. Then get the error of SDL2. Without close retropie_setup screen. I make the change mentioned here:
https://741490.bugs.gentoo.org/attachment.cgi?id=670840
Both files edited with sudo. And saved the changes. After that i go to Manage dependency packages, sdl2, install from source. And was sucefully compiled. After that go to install emulationstation and retroarch in main packages and now compile with sucess. And finally mupen64plus that i see failed too. I hope this help to the developers genius. Thanks for make Retropie

Luciano2018, 2022-02-15:
Closed because is not a real issue, just info to fix something.

closed 2022-02-15 by Luciano2018

darksaviorx, 2022-02-18:
This fix is not needed. You can change the sdl branch to 2.0.20 and it will compile in bullseye.

tavuntu, 2022-02-28:
Noob question: how can I change the SDL branch? is it something I can change within the RetroPie setup repo?

darksaviorx, 2022-02-28:
Edit this: https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/sdl2.sh
Change `echo 2.0.10` to `echo 2.0.20`.