RetroPie/RetroPie-Setup #3487: Libretro binaries are not available anymore

tavuntu, 2022-02-22:
Using lastest dietpi OS (Raspberry Pi 3) and trying to do some manual installation of libretro cores using binaries.

Tried:

```shell
./retropie_packages.sh lr-fceum install_bin
./retropie_packages.sh lr-snes9x install_bin
./retropie_packages.sh lr-fbneo install_bin
./retropie_packages.sh lr-stella2014 install_bin
```

but it throws `Could not find a binary for xxxx`, what happened here? this was working alright not long ago (I've been trying on DietPi with no issues until a few months ago when this started happening).

There's something [here](https://retropie.org.uk/forum/topic/31083/manual-source-installation-fails-in-pi4-no-binary-install-available) related to this but it's related to some KMS stuff and Raspberry pi 4

cmitu, 2022-02-22:
I don't see how this is an issue with RetroPie-Setup.
Binaries are compiled on and intended to run on a RetroPie supported system - DietPi is not one of them.

Regarding your problem - the topic you found may be relevant to your issue. If you're using KMS on your system, then there are no binaries provided for PI3, only for Pi4.

closed 2022-02-22 by cmitu

tavuntu, 2022-02-22:
It's just weird that this was working without any issues not long ago. It makes sense to me that DietPi is not officially supported tho so I'll keep looking. Thanks for the quick response.

tavuntu, 2022-02-26:
Just as an update, even fKMS doesn't seem to be working, I get a weird error while trying to compile the `libsdl2-dev (custom)` package (this happens when running the `retropie_setup.sh` for the first time. There's some more context [in here](https://github.com/MichaIng/DietPi/issues/1653) (specially for the last comments).

joolswills, 2022-02-27:
Bullseye isn't supported yet.

tavuntu, 2022-05-20:
@joolswills is there any tentative date on when (or if) it'll be supported?

tavuntu, 2022-05-21:
My bad, I probably got confused with some other repos.