RetroPie/RetroPie-Setup #3499: SDL2 build fails while linking SDL_wayland*.c due undefined symbols:

twojstaryzdomu, 2022-03-15:
SDL2 build fails while linking SDL_wayland*.c due undefined symbols:

```
Version: 4.7.21 - Last Commit: 5 days ago (5255c5b)
System: rpi3 (armv7l) - running on Raspbian GNU/Linux bookworm/sid
```

(sorted output)
```
SDL_waylanddatamanager.c:(.text+0x6c0): undefined reference to `wl_proxy_get_version'
SDL_waylanddatamanager.c:(.text+0x764): undefined reference to `wl_proxy_get_version'
SDL_waylanddatamanager.c:(.text+0x8cc): undefined reference to `wl_proxy_get_version'
SDL_waylanddatamanager.c:(.text+0x93c): undefined reference to `wl_proxy_get_version'
SDL_waylanddatamanager.c:(.text+0x9e0): undefined reference to `wl_proxy_get_version'
SDL_waylanddatamanager.c:(.text+0xb7c): undefined reference to `wl_proxy_get_version'
SDL_waylandevents.c:(.text+0x10b8): undefined reference to `wl_proxy_get_version'
SDL_waylandevents.c:(.text+0x1fc): undefined reference to `wl_proxy_get_version'
SDL_waylandevents.c:(.text+0x8fc): undefined reference to `wl_proxy_get_version'
SDL_waylandevents.c:(.text+0xdc8): undefined reference to `wl_proxy_get_version'
/usr/bin/ld: build/.libs/SDL_waylanddatamanager.o: in function `Wayland_data_device_clear_selection':
/usr/bin/ld: build/.libs/SDL_waylanddatamanager.o: in function `Wayland_data_device_set_selection':
/usr/bin/ld: build/.libs/SDL_waylanddatamanager.o: in function `Wayland_data_device_set_serial':
/usr/bin/ld: build/.libs/SDL_waylanddatamanager.o: in function `Wayland_data_offer_destroy':
/usr/bin/ld: build/.libs/SDL_waylanddatamanager.o: in function `Wayland_data_offer_receive':
/usr/bin/ld: build/.libs/SDL_waylanddatamanager.o: in function `Wayland_data_source_destroy':
/usr/bin/ld: build/.libs/SDL_waylandevents.o: in function `data_device_handle_enter':
/usr/bin/ld: build/.libs/SDL_waylandevents.o: in function `pointer_handle_button':
/usr/bin/ld: build/.libs/SDL_waylandevents.o: in function `seat_handle_capabilities':
/usr/bin/ld: build/.libs/SDL_waylandevents.o: in function `Wayland_data_source_create':
/usr/bin/ld: SDL_waylanddatamanager.c:(.text+0x6d8): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylanddatamanager.c:(.text+0x788): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylanddatamanager.c:(.text+0x8e8): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylanddatamanager.c:(.text+0x95c): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylanddatamanager.c:(.text+0x9fc): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylanddatamanager.c:(.text+0xa50): undefined reference to `wl_proxy_get_version'
/usr/bin/ld: SDL_waylanddatamanager.c:(.text+0xa68): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylanddatamanager.c:(.text+0xaf4): undefined reference to `wl_proxy_get_version'
/usr/bin/ld: SDL_waylanddatamanager.c:(.text+0xb14): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylanddatamanager.c:(.text+0xb9c): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylandevents.c:(.text+0x220): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylandevents.c:(.text+0x268): undefined reference to `wl_proxy_get_version'
/usr/bin/ld: SDL_waylandevents.c:(.text+0x288): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylandevents.c:(.text+0x2e8): undefined reference to `wl_proxy_get_version'
/usr/bin/ld: SDL_waylandevents.c:(.text+0x30c): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylandevents.c:(.text+0x91c): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylandevents.c:(.text+0x92c): undefined reference to `wl_proxy_get_version'
/usr/bin/ld: SDL_waylandevents.c:(.text+0x954): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylandevents.c:(.text+0x95c): undefined reference to `wl_proxy_get_version'
/usr/bin/ld: SDL_waylandevents.c:(.text+0x97c): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylandevents.c:(.text+0xde8): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylandevents.c:(.text+0xeb8): undefined reference to `wl_proxy_get_version'
/usr/bin/ld: SDL_waylandevents.c:(.text+0xedc): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylandevents.c:(.text+0xf20): undefined reference to `wl_proxy_get_version'
/usr/bin/ld: SDL_waylandevents.c:(.text+0xf40): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylandevents.c:(.text+0xf5c): undefined reference to `wl_proxy_get_version'
/usr/bin/ld: SDL_waylandevents.c:(.text+0xf78): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylandevents.c:(.text+0xf98): undefined reference to `wl_proxy_get_version'
/usr/bin/ld: SDL_waylandevents.c:(.text+0xfbc): undefined reference to `wl_proxy_marshal_flags'
/usr/bin/ld: SDL_waylandevents.c:(.text+0xfcc): undefined reference to `wl_proxy_get_version'
/usr/bin/ld: SDL_waylandevents.c:(.text+0xfec): undefined reference to `wl_proxy_marshal_flags'
```

```
ii  libva-wayland2:armhf                 2.14.0-1                           armhf        Video Acceleration (VA) API for Linux -- Wayland runtime
ii  libwayland-bin                       1.20.0-1                           armhf        wayland compositor infrastructure - binary utilities
ii  libwayland-client++0:armhf           0.2.8-2                            armhf        wayland compositor infrastructure - client library C++ bindings
ii  libwayland-client0:armhf             1.20.0-1                           armhf        wayland compositor infrastructure - client library
ii  libwayland-cursor++0:armhf           0.2.8-2                            armhf        wayland compositor infrastructure - cursor library C++ bindings
ii  libwayland-cursor0:armhf             1.20.0-1                           armhf        wayland compositor infrastructure - cursor library
ii  libwayland-dev:armhf                 1.20.0-1                           armhf        wayland compositor infrastructure - development files
ii  libwayland-egl++0:armhf              0.2.8-2                            armhf        wayland compositor infrastructure - EGL library C++ bindings
ii  libwayland-egl1:armhf                1.20.0-1                           armhf        wayland compositor infrastructure - EGL library
ii  libwayland-server0:armhf             1.20.0-1                           armhf        wayland compositor infrastructure - server library
ii  wayland-protocols                    1.25-1                             all          wayland compositor protocols

```

cmitu, 2022-03-15:
RaspiOS Bullseye is not supported at the moment - SDL2 is just one of the failing packages.

closed 2022-03-15 by cmitu

twojstaryzdomu, 2022-03-15:
So what is precisely missing? This has been closed all too quickly and the obvious response is not hepful here. There is a reason I'm not posting in the forums.

What would it take to enable those missing symbols? I see some compat changes for wayland 1.20, have you seen [this](https://github.com/libsdl-org/SDL/commit/e2ade2bfc46d915cd306c63c830b81d800b2575f#diff-03f611e039cfbce9094bdf1c8145f0c1c540b9d530b3e699cee9bf02da37c426)?

yllekz, 2022-05-27:
This is still broken....Why is this issue closed?