RetroPie/RetroPie-Setup #3503: Standalone MAME cursor bug / LR-MAME not saving high scores.

Dannonew, 2022-03-19:
So I've noticed this weird bug when using current branch MAME standalone on Retropie, on both a RPi 4b and an RPi 3b+. This has been the case in all versions from MAME .224 through the current version .241.

Any game you launch using the standalone MAME emulator in REtropie, a mouse cursor appears in the top left corner of the screen right when you launch it, regardless of whether you have a mouse/trackball/spinner or none of the above. It even happens when the only connected controller is a digital keyboard.

It only appears for a few seconds and then disappears while the game initializes, but I'm trying to get my Arcade Cab to behave as faithful as possible, and none of these classic arcade games would ever display a mouse cursor, so it's kind of distracting. Also - if you run it on an RPi that _does_ have a mouse or trackball/analog spinner attached, and you move the mouse/trackball when it appears, the cursor moves 1/4 of the way across the screen and then freezes, and it stays there the entire time you're playing any game and never goes away, even if the game in question has no analog controls. So if you accidentally bump an analog controller while a game is starting, you get stuck with a mouse cursor for the whole game.

LR-MAME (current branch in Libretro) actually runs perfectly like the standalone MAME, and doesn't have the cursor bug described above with one exception: it doesn't save high scores. High Score saving works fine in standalone current branch MAME but doesn't in LR-MAME.

Is there any chance either of these can be looked at? I realize standalone MAME is considered an "experimental" Emulator on Retropie, but it's been in there for a couple of years now and this bug persists.

Or - alternately, if high score saving could be implemented in LR-MAME, it would essentially fix the problem, as that one doesn't display the cursor bug.

joolswills, 2022-03-19:
You should open a topic on the forum first as detailed when opening an issue here.

Dannonew, 2022-03-19:
I did. A week ago. 30+ views and no replies.

cmitu, 2022-03-24:
Discussion continued in the forum.

closed 2022-03-24 by cmitu

Dannonew, 2022-03-24:
LOL, why was this issue closed? The bug still exists, and the "discussion" in the forum consisted of a FBNeo Dev trying to convince me to use FBNeo instead. This is a Retropie Bug with MAME standalone.