RetroPie/RetroPie-Setup #3553: script relies on proprietary DNS servers which are blocked in my area

twojstaryzdomu, 2022-06-15:
I'm having trouble resolving the internet-facing IP from some network on occasion.

It may be the doing of some network providers, in an attempt to drive DNS traffic off of `8.8.8.8` to their own network's DNS. In this instance it causes some functions that depend on `getIPAddress` to be broken.

I enclose PR #3552 that performs a more resilient route lookup.

cmitu, 2022-06-15:
Don't open additional issues for the same problem - you've already opened a PR explaining what you think it needs correction.

twojstaryzdomu, 2022-06-15:
So if I posted the issue and the fix in reverse order would you be ok? Anyway I hope this problem is resolved.

cmitu, 2022-06-15:
There's no need to duplicate the info - this is not paid work and it's all done voluntarily. If there's something to be fixed, either a forum post (preferably) or isse/PR is ok.

twojstaryzdomu, 2022-06-15:
Contrary to your understanding, a PR is NOT an issue so nothing is duplicated.  The issue is opened by way of documenting a problem. The PR is merely one solution to the issue; there may be others.

Please do not close the issue since it is still an outstanding problem with relying on a proprietary set of DNS hosts. By closing the issue prematurely (and without an actual solution in place) you're not really allowing other users to view it and confirm if they're not also facing it.