RetroPie/RetroPie-Setup #3554: runcommand.sh: does not capture logs when `/dev/shm` is unavailable or restricted

twojstaryzdomu, 2022-06-15:
`/dev/shm` isn't traditionally the location to dump data to. On some environments it may not be available (it is after all an in-memory shared location). Please enable logs to be written elsewhere (`/tmp`) if `/dev/shm` isn't available.

cmitu, 2022-06-15:
This is by design - a memory backed location was chosen in order to prevent writing to the sdcard on the Pi (the primary usage for RetroPie). 
From the systems that RetroPie supports, is there a distro where `/dev/shm` is not available and writable ?

twojstaryzdomu, 2022-06-15:
> 
I do not question the intent. It's all very well and in fact what most users want, not to write to flash too often.

I question the code - do you need to hard-code /dev/shm every time in your script? Why not define it via a variable and set it at the top of the script, enabing an easy change to a different location.

Let me give you another scenario: my game crashes the box and I would like it to capture output to disk. Nothing you write to memory would be preserved across reboots. Do you understand this case?

twojstaryzdomu, 2022-06-16:
> I think it would just be simpler to allow this to be set in the runcommand configuration file - your change is just hardcoding more locations. I don't think it's a big issue though. I can make a suitable change.

Very well but you haven't got rid of the hard-coded `/dev/shm` that are all over the place in `runcommand.sh`.

joolswills, 2022-06-16:
> 

I missed two references which I will fix.

twojstaryzdomu, 2022-06-16:
> > 
> 
> I missed two references which I will fix.

Thanks! Please see #3557, I think it's only fair that it's a value that a user may modify easily via the menu like the others.