RetroPie/RetroPie-Setup #3571: lr-mess update from source fails due to new .so filename

Widge-5, 2022-08-14:
Lately lr-mess has failed to update from source ( I am using a Raspberry Pi 4).  I have been able to successfully compile earlier releases but this one failed. I tried on two Pi's and a friend also tried on another multiple times.

The log seemed to indicate that the compile was successful, but the .so transfer failed.

> Linking mamemess_libretro.so...
> Removing additional swap
> Could not successfully build lr-mess - MESS emulator - MESS Port for libretro (/home/pi/RetroPie-Setup/tmp/build/lr-mess/mess_libretro.so not found).

The clue here is the first line quoted where it states it is attempting to link a file called _mamemess_libretro.so_ but then on the third line it says it can't find a file called _mess_libretro.so_.
Until now, the .so file for mess has been called _mess_libretro.so_ and that is the filename referenced in the lr-mess install script at `/home/pi/RetroPie-Setup/scriptmodules/libretrocores/lr-mess.sh`. It seems the filename for the mess .so file has been changed with the latest update.
Checking the tmp/build folder, I find a fully compiled file called _mamemess_libretro.so_ confirming that the build completed but the process failed because the filename expected wasn't found.

I propose updating the script to replace all instances of _mess_libretro.so_ with _mamemess_libretro.so_ (of which I think there are 3).  I tried it on my own pi and the update from source successfully completed and updated all relevant entries in emulators.cfg with the new filename.

stuken, 2022-08-14:
This was actually raised in a merge comment on the libretro repo so i thought it should be commented on here.
This is an upstream change caused by a refactor of the mame makefile generation scripts. Prior to 0.245 there was a specific mess target script. For 0.245 this was removed in favour of dynamically generating it. The output binary name is now created based off the target and subtarget values. You can verify this by building upstream with "SUBTARGET=mess" and you will find the build artifact will be "mamemess". The libretro specific changes simply append "_libretro" to the make target value. I think this will be default behaviour for upstream for the forseeable future as it unifies the naming scheme for other subtarget builds with previously followed this target/subtarget scheme.

closed 2022-08-14 by joolswills

cmitu, 2022-08-14:
@stuken @Widge-5 thank you for the heads up.

PR #3572 (2022-08-14): lr-mess: update the binary core target result
Recent changes in upstream MAME changed the resulting target filename, consequently the libretro core file needs to be renamed.

Fixes: #3571