RetroPie/RetroPie-Setup #3617: lr-scummvm experimental package fails to build (x86)

bingo1105, 2023-01-15:
It looks like there is some active development on this package, as the build script references a git "master" branch that no longer exists. There is, however, a "main" branch. I tried to modify the script to reference this new branch but was still unsuccessful in getting it to compile. I'm guessing this is a simple fix...

cmitu, 2023-01-15:
If fails to build on all platforms, no just for PC.

The main repository has been re-organized and the build process changed entirely. I was waiting for the build to stabilize before changing the scriptmodule. Maybe the time is now.

closed 2023-01-16 by joolswills

cmitu, 2023-01-16:
@bingo1105 can you update your RetroPie-Setup script and try building again the ScummVM libretro core ? The build should work now.

bingo1105, 2023-01-16:
@cmitu Updated my installation, package built successfully. Thanks very much for the quick turnaround...

cmitu, 2023-01-16:
Upload the last part of the build log on pastebin.com and post the link.

bingo1105, 2023-01-16:
@cmitu The relevant part of the log output should be here:
https://pastebin.com/9dLD0eMM

Everything above that point appears normal to me. If you need more logging detail, please let me know.

cmitu, 2023-01-17:
I don't get that failure, so there's probably something that's missing on your system - `zip`, or `wget` ?
Can you run:
```
cd $HOME/RetroPie-Setup/tmp/build/lr-scummvm
sudo ./bundle_datafiles.sh bundle scummvm.zip
```
and post the output ?

bingo1105, 2023-01-17:
@cmitu Ugh, you're right, missing zip package. Thank you!

cmitu, 2023-01-17:
Yes, I assumed as much. Seems we don't add `zip` as a base requirements for RetroPie (just `unzip`).
We'll add it as a dependency, thanks for testing.

PR #3619 (2023-01-16): lr-scummvm: update build recipe, upgrade the core to v2.6.1
Recently, upstream re-organized the repository and changed the build process. This commit adapts the scriptmodule to the new structure:

 - building is done directly from the main repo source
 - data files neeed for the core is built with `make datafiles` and copied to our package
 - ScummVM is upgraded to current stable (v2.6.1) in the process. NOTE: WIP engines are not built by default and are not included
 - switched the theme the `scummremastered`, added in 2019 in ScummVM
 - license changed to GPLv3 (same as the main ScummVM proj [... cut]