RetroPie/RetroPie-Setup #3650: Adjust defaults for lr-beetle-pce-fast / lr-beetle-pce

joolswills, 2023-03-01:
Need to test the new module lr-beetle-pce on slower devices (rpi/rpi3), and make changes to:

Either:
 * Change which is the default emulator based on platform/cpu
 * Change section depending on platform/cpu (eg - keep pce-fast in main for slower devices, but in optional for faster devices and keep pce in main for faster devices, but in optional for slower).

joolswills, 2023-03-01:
@nemo93 Do you have any recommendations for games I should test the lr-beetle-pce core with on slower devices? I assume some games may run better than others - ideally something that would show slowdowns on a slower device.

nemo93, 2023-03-02:
@joolswills try first the [Arcade CD-ROM² titles](https://necretro.org/CD-ROM%C2%B2_systems#Arcade_CD-ROM.C2.B2). I'd say go first with:

* Ryuuko no Ken (CD)
* Garou Densetsu Special (CD)
* Ginga Fukei Densetsu Sapphire (CD)
* Street Fighter II': Champion Edition (PCE)
* Ghouls 'n Ghosts (Dai Makaimura, SuperGrafx)
* 1941 (SuperGrafx)

This should get you covered.

joolswills, 2023-03-04:
@nemo93 Thank you.

cmitu, 2024-06-29:
FWIW I tested some of the games recommended by @nemo93 recently with the `lr-beetle-pce` on a Pi4 and didn't see any slowdowns. Maybe it's time we make it default on Pi4 and up ?