RetroPie/RetroPie-Setup #3693: Add Vanilla Conquer C&C engine

gizmo98, 2023-03-21:
Vanilla Conquer is a C&C engine which can run 
Tiberian Dawn and Red Alert. https://github.com/TheAssemblyArmada/Vanilla-Conquer

Exarkuniv, 2023-03-27:
@gizmo98

s1eve-mcdichae1, 2024-01-04:
WIP: https://gist.github.com/s1eve-mcdichae1/90edfe26d735a1b6e021ce523ee52e0f

I think this has improved file handling vs. the current RP-Extra module.

Installed executables are kept separate from user-provided game assets (or downloaded demo), and symlinks are used to combine them all in the install directory.

A unified launcher script allows for cleaner config dirs.

$home/.config/vanilla-conquer dir linked to $md_conf_root/vanillacc for network access and backup of savegames and configs.