RetroPie/RetroPie-Setup #3721: update attractmode the source

22marco77, 2023-05-27:
Pi model or other hardware: pi 3 B+,
power supply used: universal raspberry power supply
used RetroPie version 4.8.4 buster
built from: downloaded image from retropie official site, installation from scratch
connected USB devices: none
controller used:
DragonRise error messages received:
Detailed log (if applicable):
Log started at: Wed 24 May 2023, 09.15.45, CEST

RetroPie-Setup version: 4.8.4 (0bcf01d4)
System: rpi3 (armv7l) - Raspbian GNU/Linux 10 (buster) - Linux retropie 5.10.103-v7+ #1529 SMP Tue Mar 8

could not successfully build attractmode - attractmode  emulator fronted
(/home/pi/RetroPie-Setup/tmp/build/attractmode/attract/attract not found

this is my directory
/home/pi/RetroPie-Setup/tmp/build/attractmode/attract

cmitu, 2023-05-27:
You already have a forum topic opened at https://retropie.org.uk/forum/topic/34259/ - there's no need to open a separate issue and you can add there any additional info for troubleshooting.

closed 2023-05-27 by cmitu

22marco77, 2023-05-27:
Ok, I'm sorry. I will wait for the solution if someone can find it, I updated retropie setup and then I started the update but this solution did not work for me