RetroPie/RetroPie-Setup #3748: scummvm unable to change extrapath

voggur, 2023-09-08:
I had alot of confusion why extrapath parameter was always resetting to the default parameter.
In scummvm.sh this is hard-coded will causes scummvm to reset this parameter.
My problems was solved be editing the scummvm.sh
`$md_inst/bin/scummvm --fullscreen --joystick=0 --extrapath="$md_inst/extra" "\$game"`
to
`$md_inst/bin/scummvm --fullscreen --joystick=0 "\$game"`
I suggest updating this coed as this is already set in scummvm.ini and should not be necessary and could cause user confusion.

https://github.com/RetroPie/RetroPie-Setup/blob/5f2b85a70640b34a547d3b2e51716d5752bef604/scriptmodules/emulators/scummvm.sh#L89C48-L89C48

voggur, 2023-09-08:
I have my folder sett to "~/RetroPie/roms/scummvm/scummvm-extra/" 
1. for ease access to add files like MT-32 emulation and soundfont.
2. Some of this files like .sf2 can be large and "roms" directory is smb mounted to my file-server.

cmitu, 2023-09-18:
@voggur we removed the hardcoded `--extrapath` in #3751, it should make it easier for user to choose their own folder without being overridden by `scummvm`'s start-up script.

closed 2023-09-18 by cmitu