RetroPie/RetroPie-Setup #3894: Retroarch home directory symlink

9p6, 2024-03-29:
The current retropie setup hijacks `~/.config/retroarch` directory from existing installation by symlinking it to `/opt/retropie/configs/all/retroarch`. This makes it awkward to run an independent retroarch installation  (installed from Debian repo, Ubuntu PPA, compiled from source...)  because now two different versions share the same config dir. It can also lead to data loss during removal for anyone unaware of implications of this setup.

What significance does the symlink have for retropie? Does it exist solely for convenience of someone running `/opt/retropie/emulators/retroarch/bin/retroarch` directly? Is it necessary to support it in such way?

cmitu, 2024-03-30:
> The current retropie setup hijacks `~/.config/retroarch` directory from existing installation by symlinking it to `/opt/retropie/configs/all/retroarch`. This makes it awkward to run an independent [...]

Yes, that's true and the same it's true for any other program/emulator/port that's handled by RetroPie - after all, that's what is suppose to happen. The programs installed by RetroPie are 'isolated', the scripts will just automate their installation, 

> It can also lead to data loss during removal for anyone unaware of implications of this setup.

 Yes and this si why the removal has a big warning about keeping or saving the configuration, specifically for this cases.

> 
> What significance does the symlink have for retropie? Does it exist solely for convenience of someone running `/opt/retropie/emulators/retroarch/bin/retroarch` directly? Is it necessary to support it in such way?

The `config` folder is shared over the network in order to be easy to back-up or modify/copy files to it, just like the `roms` or `bios` folders.

If you have any other support questions, please use the forum, as instructed in the issue template.

closed 2024-03-30 by cmitu

9p6, 2024-04-01:
> Yes, that's true and the same it's true for any other program/emulator/port that's handled by RetroPie - after all, that's what is supposed to happen.

How so? Each of these programs has own configuration stored in `/opt/retropie/configs`. And setup is free to write them as needed:

https://github.com/RetroPie/RetroPie-Setup/compare/master...9p6:RetroPie-Setup:master

Retroarch was a bit tricky to get because it has so many subdirectories but for other packages it should be a lot easier.

> The programs installed by RetroPie are not 'isolated', the scripts will just automate their installation.

The expectation is that add-on software installed to `/opt` doesn't interfere with regular system software. And a lot of software in retropie is part of Debian (Retroarch, Dolphin, Dosbox).

> The config folder is shared over the network in order to be easy to back-up or modify/copy files to it

Sure the config folder can still link to retropie configurations without seizing default config locations.

> If you have any other support questions

This is not a support question