RetroPie/RetroPie-Setup #3898: Ubuntu or General Desktop Environment - Prevent Double Istance of "Retropie"... it's possible ?

DjDiabolik, 2024-04-02:
[image]

At default Ubuntu GNOME Desktop on Left Bar it's appear the icon "Rpie" (because Rpie and not Retropie ?!!? I can edit it ? How i can do ? But it's a totally different question..lol).

From GNOME Desktop it's need a single click and a Gnome-Terminal has been launched and after EmulationStation.
if you double click by mistake two difference istance of ES it's been started and this creates some general problems.

Can i prevent this appends? It's GNOME have some setting to prevent this ?

cmitu, 2024-04-02:
Please use the forum for general questions.

closed 2024-04-02 by cmitu

DjDiabolik, 2024-04-02:
ops..
You thinks it's better to use "Help and Support" or "Ideas and Development" ? @cmitu