RetroPie/RetroPie-Setup #3905: Update amiberry source to latest

n0rt0nthec4t, 2024-04-22:
Included version of amiberry is several releases behind (included is 5.6.6, current is 5.7.0)

cmitu, 2024-04-22:
Please use the forum for support questions and suggestions. The 5.7.0 was released less than 24 hours and we don't bump versions on certain emulators without testing it first.

closed 2024-04-22 by cmitu

n0rt0nthec4t, 2024-04-22:
Where was 5.6.7 and 5.6.8 then? They have important bug fixes for Amiga emulation

joolswills, 2024-04-22:
You're welcome to submit PRs. We work on this in our free time. As @cmitu mitu said you can use the forum to bring these things to our attention.

cmitu, 2024-04-22:
@joolswills I actually have a branch for it ready for 3 weeks (re-did out patches for [5.7.0](https://github.com/cmitu/RetroPie-Setup/tree/amiberry-5.7.0)), just need to re-test it again.