RetroPie/RetroPie-Setup #3915: retroarch failing to build on clean ubuntu noble install

theofficialgman, 2024-05-02:
```
input/drivers/udev_input.c:3855:14: warning: the comparison will always evaluate as ‘true’ for the pointer operand in ‘binds + (sizetype)((long unsigned int)port * 4272)’ must not be NULL [-Waddress]
 3855 |          if (binds[port])
      |              ^~~~~
CC gfx/drivers_context/gfx_null_ctx.c
CC gfx/common/gl_common.c
CC gfx/drivers/gl2.c
CC libretro-common/gfx/gl_capabilities.c
CC libretro-common/glsym/rglgen.c
CC gfx/drivers/gl1.c
CC gfx/drivers_context/vc_egl_ctx.c
gfx/drivers_context/vc_egl_ctx.c:24:10: fatal error: VG/openvg.h: No such file or directory
   24 | #include <VG/openvg.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:209: obj-unix/release/gfx/drivers_context/vc_egl_ctx.o] Error 1
make: *** Waiting for unfinished jobs....
Could not successfully build retroarch - RetroArch - frontend to the libretro emulator cores - required by all lr-* emulators (/home/runner/RetroPie-Setup/tmp/build/retroarch/retroarch not found).
```

vc_egl shouldn't be even attempted to be built on non-raspberry pi systems. this is probably a mistake in the configuration parameters.

full buildlog https://github.com/cobalt2727/L4T-Megascript/actions/runs/8923983609/job/24509379849#step:3:2716

cmitu, 2024-05-03:
We don't set any configuration parameters for OpenVG, they're set [auto](https://github.com/libretro/RetroArch/blob/ab04f8a7a248f4d96ad3d55561b5e4d40f38c1ca/qb/config.params.sh#L109) in the `configure` scripts. I don't think this is a RetroPie build issue, you may want to open this upstream or explicitely disable the `openvg` support in your scripts.

mrcmunir, 2024-05-22:
> ```
> input/drivers/udev_input.c:3855:14: warning: the comparison will always evaluate as ‘true’ for the pointer operand in ‘binds + (sizetype)((long unsigned int)port * 4272)’ must not be NULL [...]

The workarround it's use isPlatform on all tegra devices  depends+=(--disable-videocore --disable-vg) in retroarch.sh script .
It's broken by retroarch for a long time and i'm compile fine with force these flags

cmitu, 2024-05-23:
> It's broken by retroarch for a long time and i'm compile fine with force these flags

Unless you report it upstream, the issue is not going to be fixed as is.

theofficialgman, 2024-05-24:
> The workarround it's use isPlatform on all tegra devices params+=(--disable-videocore --disable-vg) in retroarch.sh script . It's broken by retroarch for a long time and i'm compile fine with force these flags

The CI I linked doesn't have device-tree available (as its a QEMU emulated chroot), so its just using the fallback platform provided here https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/system.sh#L490 not any of the tegra specific flags.

theofficialgman, 2024-05-24:
> @theofficialgman try [mrcmunir/RetroArch@0731ef8](https://github.com/mrcmunir/RetroArch/commit/0731ef8c7b48be707ef5fa7cd75135bc7132cd0f) If it's prevent compile vc_egl headers without pass disabled flags.

I don't think that is the correct course of action. You should go through the logs and determine what is pulling in compilation of that file even when when OPEN_VG gets disabled from auto https://github.com/cobalt2727/L4T-Megascript/actions/runs/8923983609/job/24509379849#step:3:2763

mrcmunir, 2024-05-25:
@theofficialgman Probably found cause fail retroarch bcm_host_init fuction are found.
https://github.com/cobalt2727/L4T-Megascript/actions/runs/8923983609/job/24509379849#step:3:2729

cmitu, 2024-06-10:
I'm going to close this, since it's not caused by RetroPie's build options - either talk to upstream to fix the problem or provide a fix for it. On a PC Ubuntu 24.04 system the build works fine, so it's not an Ubuntu 24.04 general issue, it's platform specific (ARM/aarch64).

closed 2024-06-10 by cmitu