RetroPie/RetroPie-Setup #3925: Issue with lr-scummvm hardware acceleration on rpi4

dszakallas, 2024-05-28:
ScummVM recently added OpenGL(ES) acceleration and support is auto-detected at compile time. However I suppose there's an issue with the detection itself as it requests OpenGL context, which is not available.

```
[ERROR] Requesting OpenGL context, but RetroArch is compiled against OpenGLES. Cannot use HW context.
[ERROR] [Environ]: SET_HW_RENDER - Dynamic request HW context failed.
[libretro WARN] Failed to set up hardware rendering, falling back to software.
```

This behavior is not really problematic, but results in an annoying pop-up `Failed to set HW rendering` on game launch.

cmitu, 2024-05-29:
Thanks for the heads up. FWIW, there's no 'compile time detection', support for one or the other GL flavor is specified with an option, I guess OpenGL is selected as default. We'll adjust the compilation flags and see if this warning disappears.

In the future, please use the forum for reports, as indicated in the issue template.

cmitu, 2024-05-29:
The _Makefile_ pre-sets the `HAVE_OPENGL[ES2]` flags depending on the `platform` parameter, ignoring any `HAVE_OPENGL/OPENGLES2` parameters passed to the `make` command. Using:
```
make platform=rpi4_64
```
will force OpenGLES2. I don't see how - besides patching the _Makefile_ - a hardware renderer choice can be passed on. We'll have to probably talk to upstream about this.

cmitu, 2024-05-31:
@dszakallas this should be fixed with the recent changes in #3927. Can you update and give it a try ?

dszakallas, 2024-06-03:
Seems to work for me. No popup and HW acceleration is now on in Video Settings. Thanks!

closed 2024-06-03 by dszakallas