RetroPie/RetroPie-Setup #3937: Make Amiberry available for x86_64 as well

midwan, 2024-06-20:
Is there a reason why Amiberry is not enabled/available for x86_64 platforms a well?
It runs quite well there for a few years now, so perhaps this was missed?

The only difference when compiling from source would be the PLATFORM option (instead of `rpi3`, use `x86-64`) in the Makefile. 
And I have pre-built binaries for releases as well, even though I know you prefer your own binaries. :)

midwan, 2024-06-20:
In fact, on x86-64 you could use the newer and more compatible `preview` branch, which contains v6.x
It's a bit more demanding in CPU power, which is why it hasn't completely replaced v5.x yet. But that's not an issue for x86, only for RPI3/RPI4 (RPI5 is fast enough to run everything full speed even in v6.x).

midwan, 2024-06-20:
Note: if you want to avoid the whole PLATFORM=XYZ thing, you could alternatively use CMake to build it, also.
It will automatically detect between Linux or macOS (though I'm guessing you don't have a macOS RetroPie installation), and build using the native arch anyway. You could still probably get more optimized binaries with the specific CPU flags, however.

cmitu, 2024-06-21:
> Is there a reason why Amiberry is not enabled/available for x86_64 platforms a well?
> It runs quite well there for a few years now, so perhaps this was missed?

 Yes, the addition of the PC platform in Amiberry wasn't reflected in our scriptmodule. I guess that either there's no much demand for Amiga emulation on x86-64 or the existing users are content with FS-UAE, 
 I'm also aware of the `preview` / `v6` branch, but since Amiberry we target ARM/SBCs, we're a bit under-powered for that. I guess we can enable Amiberry for PC archs also - I guess I'll add my comments on #3938.

commit 66012344 (2024-06-28): Merge pull request #3938 from BlitterStudio/add-amiberry-x86

Add Amiberry for x86 platforms as well (fixes #3937)
closed 2024-06-28 by joolswills