RetroPie/RetroPie-Setup #3963: Amiberry no longer runs on RetroPie

kec2, 2024-08-02:
Amiberry does not start after update on retropie.

RetroPie v4.8.8
Amiberry v 5.7.1
Raspbarry pi 4

content of /dev/shm/runcommand.log
```
Parameters: 
Executing: bash "/home/pi/RetroPie/roms/amiga/+Start Amiberry.sh"
Launching ...
./amiberry
malloc(): unsorted double linked list corrupted
/opt/retropie/emulators/amiberry/amiberry.sh: linje 107:   759 Afbrudt (SIGABRT)       "$emulator" "${params[@]}"
~
```

I have had a chat with BlitterStudio and it seams that RetroPie is the couse.
[https://github.com/BlitterStudio/amiberry/issues/1392](https://github.com/BlitterStudio/amiberry/issues/1392)

cmitu, 2024-08-02:
Please use the forum for support issues, as explained in the issue template.

closed 2024-08-02 by cmitu

kec2, 2024-08-02:
This is an update of RetroPie the broke the installations of amiberry. It worked very well before. So I don't understand why you are closing this issue.

vaccinemedia, 2024-12-14:
@kec2 if you edit the amiberry script and change the version number to 5.7.4 it should work ok if you uninstall it and re-install using retropie setup