RetroPie/RetroPie-Setup #3997: SuperTuxKart - simple manual install, please add to Ports

stevenjaycohen, 2024-11-05:
Link: https://supertuxkart.net/Main_Page

**Steps**
1. Install at command line on Debian 12 using apt
2. Copy the Ports Menu Entry for SuperTux to make a new one for Super Tux Kart
3. Change the name of the executable to supertuxkart
4. Restart Emulation Station
5. **PLAY** :)

_Tested on RaspberryPi 5 over Debian 12 and RetroPie_

Since this doesn't seem to require any coding, could it be added to the Ports menu officially?

DjDiabolik, 2024-11-29:
SuperTuxKart it's also available as Flatpak...... Couldn't it be an idea to integrate flathub packages into emulationstation ?

For example recently whit my Mini PC and Ubuntu i have tryed to compile Dolphin but apparently this not support the Vulkan.
After that from FlatHub i have installed the flatpak package and the gui it's in italian and Vulkan and OpenGL it's correctly works (so even if my hardware is not able to emulate the system well due to power limitations problably).

Gemba, 2024-12-23:
@stevenjaycohen : It is already available, in the extensive RetroPie-Extra repository which can be installed alongside with the main/this RetroPie repo.

https://github.com/Exarkuniv/RetroPie-Extra/blob/master/scriptmodules/ports/supertuxkart.sh

stevenjaycohen, 2024-12-23:
> @stevenjaycohen : It is already available, in the extensive RetroPie-Extra repository which can be installed alongside with the main/this RetroPie repo.

@Gemba -- Go it! -- Is that particular fork of RetroPie-Extra considered "official"?

Gemba, 2024-12-24:
> > @stevenjaycohen : It is already available, in the extensive RetroPie-Extra repository which can be installed alongside with the main/this RetroPie repo.
> 
> @Gemba -- Go it! -- Is that particular fork of RetroPie-Extra considered "official"?

The main RetroPie-Setup is made this way that it allows extra repositories below `RetroPie-Setup/ext/<whatever>`.
The maintainer of RetroPie-Extra is not associated with the RetroPie-Team and it is not a fork. Think of RetroPie-Extra like a set of add-ons for RetroPie. The RetroPie-Team does not tag such repos as "official". As far as I can tell, RetroPie-Extra only provides the scriptmodule. It installs games that do not have a commercial/intellectual property license, games that do require such a license you have to add by yourself.

If you think RetroPie-Extra is too much, you can also pick the scriptmodules you want and put them manually below `RetroPie-Setup/ext/<whatever>` but keep the folder structure up to the scriptmodule you want to use, also if there is a sibling folder with the same basename copy that also.
E.g. `.../Exarkuniv/RetroPie-Extra/blob/master/scriptmodules/ports/supertuxkart.sh` would go into `/home/pi/RetroPie-Setup/ext/my_picks/scriptmodules/ports/supertuxkart.sh`)

As we are getting off-topic let's continue, if needed, in the RetroPie forum.