RetroPie/RetroPie-Setup #4029: Using the latest Amiberry/Amiberry-Lite versions, in Bookworm

midwan, 2025-01-29:
Hi all,
In order to troubleshoot an issue reported to me recently (https://github.com/BlitterStudio/amiberry/issues/1617), I went through the process of installing RetroPie on top of an RPI-OS Lite installation, running Bookworm, on an RPI5.

I wanted to document the changes I did for the latest Amiberry to run on it, in the hopes that it could help you in the migration at some point and minimize any pains. Feel free to adapt things according to your needs, of course. These steps aimed to keep things in place as much as possible, so you can certainly make it even cleaner (e.g. you don't need to install the old Amiberry first, like I did).

The newer Amiberry and Amiberry-Lite releases are packaged into DEB and RPM installers. This is the easiest way to install them, if you're running one of the supported OS versions (Buster is out of support, so this will only work for Bullseye and Bookworm). It will bring in any dependencies for you, so you don't need to do that beforehand. It can also be uninstalled or upgraded, using standard syntax (e.g. `sudo apt remove amiberry` will do what you expect).

I installed Amiberry as follows:
- Installed the old (5.7.2) Amiberry from source, using the RetroPie-Setup. This wasn't necessary, but it just helped setting things up originally, such as the symlinks to the paths, which I looked at in order to configure things in the next steps.
- Used the relevant `.deb` package (bookworm, arm64) to install Amiberry and any dependencies (flac and libenet were missing).
- Renamed the RetroPie-installed older `amiberry` binary in `/opt/retropie/emulators/amiberry`, to `amiberry.572`. Just as a precaution, to make sure that does not run by mistake.
- Edited `amiberry.sh` to run `amiberry`, instead of `./amiberry`, so it can find it from the different location the DEB installed it in (system wide).
- Then I run Amiberry from the console once, to let it create the default directory structure and `amiberry.conf` file, which is created by default under XDG_CONFIG_HOME (`~/.config/amiberry/amiberry.conf`)
- After that, I edited the `amiberry.conf` file, to change some of the paths, so they point to RetroPie's. Those were:

controllers path -> `/opt/retropie/configs/all/retroarch/autoconfig`
retroarch_config -> `/opt/retropie/configs/all/retroarch/retroarch.cfg`
rom_path -> `/home/pi/RetroPie/BIOS/amiga/`
I left the rest of them at their defaults, as there was no reason to change them for now. You might want to change the floppy_path, cdrom_path, whdload_arch_path as well, perhaps.

Saved, restarted EmulationStation, and launched `Gods_v3.2_0666.lha` from there. It correctly used the newer Amiberry to run it, and I let it go all the way into the game.

cmitu, 2025-01-29:
@midwan thanks for the steps provided, I guess we'll take this into account when adding the new version. Just for completeness, which version (of Amiberry) did you install from `.deb` ?

midwan, 2025-01-29:
@cmitu I tested things on an RPI5, which can handle both Amiberry (v7.x) and Amiberry-Lite (currently at v5.8.2). In my tests, I used Amiberry 7.

Generally speaking, any device slower than the RPI5 would be better off using Amiberry-Lite, and anything faster should probably use Amiberry. The Lite version gets cherry-picked updates, but the core emulation remains the same, to keep the performance at a good level. It's also the only version (for now) that has a working JIT for ARM/ARM64, so if that's important to you, keep that version.

Amiberry on the other hand, is based on the latest WinUAE (5.3.1, as v6.0 beta is still unstable), and contains the best emulation accuracy, more expansion cards support, and JIT for AMD64. It is heavier and requires a faster board to get full speed results, hence the recommended hardware is at least an RPI5 or higher.

You can also install both of them side by side, as they won't interfere with each other. They have different binary names, and install their dependencies in separate locations.

orbitalflower, 2026-03-29:
It does seem like an oversight that the pre-made images on the Retropie website are still using Buster; it's been out of long-term support for nearly two years now. Amiberry in particular is several versions behind (v5.7.1 -> v8.1.2).