RetroPie/RetroPie-Setup #4046: Error install lr-dolphin

RobinDadswell, 2025-03-08:
Hi All,

Running retropie on Ubuntu 24.10 on an Intel i5-7300U I am getting an error installing lr-dolphin

https://gist.github.com/RobinDadswell/5b73e9de2f69e98173b6cca0a4f9705d 

Logs are below. Apologies if this is the wrong place for the report, just signpost me to the right place :D 

Cheers in advance

Robin

cmitu, 2025-03-08:
> Logs are below. Apologies if this is the wrong place for the report, just signpost me to the right place :D

Please use the forums for reporting issues, just as the issue template suggests.

closed 2025-03-08 by cmitu

RobinDadswell, 2025-03-08:
> > Logs are below. Apologies if this is the wrong place for the report, just signpost me to the right place :D
> 
> 
> 
> Please use the forums for reporting issues, just as the issue template suggests. 

Just a heads up, the github issue template didn't load in so it didn't aim me at anything else. Happy to log on the forum instead though :)

RobinDadswell, 2025-03-09:
> > Logs are below. Apologies if this is the wrong place for the report, just signpost me to the right place :D
> 
> Please use the forums for reporting issues, just as the issue template suggests.

I have looked into this a bit deeper, it turns out that you are using the legacy Issue Template which is no longer supported and for the template to work (and show up when someone goes to create a new issue) you need to move to the new template workflow 

[image]

https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/about-issue-and-pull-request-templates

[image]