RetroPie/RetroPie-Setup #4054: Capitalisation in PS2 emulators.cfg

anakwaboe4, 2025-03-21:
OS: Debian  12 x86-64

I was wondering if this was a bug specifically for me or something that is wrong in general but when i installed the PCSX2 emulator if generated the following config:

/opt/retropie/configs/ps2/emulators.cfg

`pcsx2 = "/usr/games/pcsx2 --windowed %ROM%"
pcsx2-nogui = "/usr/games/pcsx2 --fullscreen --nogui %ROM%"
default = "pcsx2-nogui"`

this gave me the error that the program was not found. I noticed in /usr/games/ that the PCSX2 programm is capitalized so changing it to:

`pcsx2 = "/usr/games/PCSX2 --windowed %ROM%"
pcsx2-nogui = "/usr/games/PCSX2 --fullscreen --nogui %ROM%"
default = "pcsx2-nogui"`

fixed it.

This file get overwritten quite often with the old config. So i'm looking for a more permanent fix.

Now i was wondering if this is a me issue or if i can fix this in the RetroPie-Setup/scriptmodules/emulators/pcsx2.sh. 

First time using retroPI so i'm wondering if i'm wrong or this is really and issue?

cmitu, 2025-03-22:
Yes, there's an error with the emulator path, introduced in 2b58c8bcf1e2ee1. Most (PC) users are using Ubuntu and thus the PPA version is installed (which is almost up-to-date), so this went unnoticed due to the different path of the PPA package.

We'll fix this, thanks for the report.

cmitu, 2025-03-22:
@anakwaboe4 should be fixed by #4055

closed 2025-03-22 by cmitu

anakwaboe4, 2025-03-22:
thank you very much for the quick fix. And a clean solution i must admit.