RetroPie/RetroPie-Setup #4074: xboxdrv can not be installed anymore, repository missing

kalaksi, 2025-05-23:
As the title says, installation fails after it prompts for github username/password. I believe the repository was https://github.com/zerojay/xboxdrv.git.

joolswills, 2025-05-23:
That repository isn't used any more by the xboxdrv module so your RetroPie-Setup script must be out of date. Please use the forum for additional support.

kalaksi, 2025-05-23:
@joolswills that's true. I did check the script https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/xboxdrv.sh before opening this issue, but apparently misread since `rp_module_license` still referred to the old repo but `rp_module_repo` was indeed already updated. Thanks for the quick reply

joolswills, 2025-05-23:
Thanks. I have fixed the licence URL now also