RetroPie/RetroPie-Setup #4082: [retropie 4.8.9] dosbox-staging does not compile

Drag0nFly, 2025-06-28:
The latest git-sourced build of “dosbox-staging” (superior to the stock DOSBox wrt. speed and compatibility) fails to build successfully on Debian Buster; presumably due to syntax changes between gcc versions. Are there any plans to support more recent versions of gcc on retropie? (I had assumed Debian Buster was deprecated at this point, but it appears it is still in use)

Relevant portions of the errorlog during compilation stage is included below–

```
[...]
meson-0.61.5/unittests/helpers.py
meson-0.61.5/unittests/internaltests.py
meson-0.61.5/unittests/linuxcrosstests.py
meson-0.61.5/unittests/linuxliketests.py
meson-0.61.5/unittests/machinefiletests.py
meson-0.61.5/unittests/platformagnostictests.py
meson-0.61.5/unittests/pythontests.py
meson-0.61.5/unittests/rewritetests.py
meson-0.61.5/unittests/subprojectscommandtests.py
meson-0.61.5/unittests/taptests.py
meson-0.61.5/unittests/windowstests.py
/home/pi
/home/pi/RetroPie-Setup/tmp/build/dosbox-staging /home/pi

= = = = = = = = = = = = = = = = = = = = =
Building 'dosbox-staging' : modern DOS/x86 emulator focusing on ease of use
= = = = = = = = = = = = = = = = = = = = =

Directory already configured.

Just run your build command (e.g. ninja) and Meson will regenerate as necessary.
If ninja fails, run "ninja reconfigure" or "meson --reconfigure"
to force Meson to regenerate.

If build failures persist, run "meson setup --wipe" to rebuild from scratch
using the same options as passed when configuring the build.
To change option values, run "meson configure" instead.
ninja: Entering directory `/home/pi/RetroPie-Setup/tmp/build/dosbox-staging/build'
[1/245] Compiling C++ object src/misc/libmisc.a.p/fs_utils_posix.cpp.o
FAILED: src/misc/libmisc.a.p/fs_utils_posix.cpp.o 
c++ -Isrc/misc/libmisc.a.p -Isrc/misc -I../src/misc -I../include -I. -I.. -Isrc/libs/ghc -I../src/libs/ghc -Isrc/libs/loguru -I../src/libs/loguru -Isrc/libs/whereami -I../src/libs/whereami -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Wpedantic -std=c++2a -O3 -Wno-unknown-pragmas -fpch-preprocess -Wmaybe-uninitialized -Weffc++ -Wextra-semi -fstrict-aliasing -Wstrict-aliasing -fmerge-all-constants -fno-math-errno -fno-signed-zeros -fno-trapping-math -frename-registers -ffunction-sections -fdata-sections -shared-libgcc -fPIC -march=native -mfpu=neon-vfpv4 -DLOGURU_STACKTRACES=1 -DLOGURU_USE_LOCALE=0 -mcpu=cortex-a72 -mfpu=neon-fp-armv8 -O2 -D_REENTRANT -isystem/usr/include/SDL2 -isystem../subprojects/libslirp-v4.7.0/src -isystem../subprojects/libslirp-v4.7.0 -isystemsubprojects/libslirp-v4.7.0 -isystem../subprojects/iir1-1.9.3 -isystemsubprojects/iir1-1.9.3 -isystem../src/libs -isystemsrc/libs -Walloca -Wctor-dtor-privacy -Wdate-time -Wdisabled-optimization -Wduplicated-branches -Weffc++ -Wextra-semi -Wlogical-op -Wlogical-not-parentheses -Wredundant-decls -Wsizeof-pointer-div -Wstack-protector -Wstrict-null-sentinel -Wsuggest-override -Wzero-as-null-pointer-constant -Wno-format-security -MD -MQ src/misc/libmisc.a.p/fs_utils_posix.cpp.o -MF src/misc/libmisc.a.p/fs_utils_posix.cpp.o.d -o src/misc/libmisc.a.p/fs_utils_posix.cpp.o -c ../src/misc/fs_utils_posix.cpp
../src/misc/fs_utils_posix.cpp: In function ‘std::optional<FatAttributeFlags> xattr_to_fat_attribs(const string&)’:
../src/misc/fs_utils_posix.cpp:193:46: error: ‘const string’ {aka ‘const class std::__cxx11::basic_string<char>’} has no member named ‘starts_with’
  if (xattr.size() <= XattrMaxLength && xattr.starts_with("0x") &&
                                              ^~~~~~~~~~~
[2/245] Compiling C++ object src/hardware/libhardware.a.p/vmware.cpp.o
FAILED: src/hardware/libhardware.a.p/vmware.cpp.o 
c++ -Isrc/hardware/libhardware.a.p -Isrc/hardware -I../src/hardware -I../include -I. -I.. -Isrc/libs/ghc -I../src/libs/ghc -Isrc/libs/loguru -I../src/libs/loguru -Isubprojects/speexdsp-1.2.1 -I../subprojects/speexdsp-1.2.1 -Isubprojects/speexdsp-1.2.1/include -I../subprojects/speexdsp-1.2.1/include -Isubprojects/speexdsp-1.2.1/include/speex -I../subprojects/speexdsp-1.2.1/include/speex -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Wpedantic -std=c++2a -O3 -Wno-unknown-pragmas -fpch-preprocess -Wmaybe-uninitialized -Weffc++ -Wextra-semi -fstrict-aliasing -Wstrict-aliasing -fmerge-all-constants -fno-math-errno -fno-signed-zeros -fno-trapping-math -frename-registers -ffunction-sections -fdata-sections -shared-libgcc -fPIC -march=native -mfpu=neon-vfpv4 -DLOGURU_STACKTRACES=1 -DLOGURU_USE_LOCALE=0 -mcpu=cortex-a72 -mfpu=neon-fp-armv8 -O2 -D_REENTRANT -isystem/usr/include/SDL2 -isystem/usr/include/libpng16 -isystem../subprojects/libslirp-v4.7.0/src -isystem../subprojects/libslirp-v4.7.0 -isystemsubprojects/libslirp-v4.7.0 -isystem../subprojects/iir1-1.9.3 -isystemsubprojects/iir1-1.9.3 -isystem../src/libs -isystemsrc/libs -Walloca -Wctor-dtor-privacy -Wdate-time -Wdisabled-optimization -Wduplicated-branches -Weffc++ -Wextra-semi -Wlogical-op -Wlogical-not-parentheses -Wredundant-decls -Wsizeof-pointer-div -Wstack-protector -Wstrict-null-sentinel -Wsuggest-override -Wzero-as-null-pointer-constant -Wno-format-security -MD -MQ src/hardware/libhardware.a.p/vmware.cpp.o -MF src/hardware/libhardware.a.p/vmware.cpp.o.d -o src/hardware/libhardware.a.p/vmware.cpp.o -c ../src/hardware/vmware.cpp
../src/hardware/vmware.cpp: In function ‘void command_get_version()’:
../src/hardware/vmware.cpp:83:24: error: ‘class std::set<std::__cxx11::basic_string<char> >’ has no member named ‘contains’
  if (!SegmentBlackList.contains(program_segment_name)) {
                        ^~~~~~~~
[3/245] Compiling C++ object src/misc/libmisc.a.p/fs_utils.cpp.o
[4/245] Compiling C++ object src/misc/libmisc.a.p/ethernet_slirp.cpp.o
ninja: build stopped: subcommand failed.
Could not successfully build dosbox-staging - modern DOS/x86 emulator focusing on ease of use (/home/pi/RetroPie-Setup/tmp/build/dosbox-staging/build/dosbox not found).
/home/pi

Log ended at: Sat 28 Jun 22:43:51 NZST 2025
Total running time: 0 hours, 0 mins, 15 secs
```

joolswills, 2025-06-28:
You can install RetroPie manually on newer versions of Raspberry Pi OS. Eg bullseye or bookworm based. Or you can use another dosbox emulator. Please use the forum for further help with that.

Drag0nFly, 2025-06-28:
Thanks for the quick reply @joolswills Yes, I am aware that I can compile it on a (much) newer OS distro, but the issue is that all my retro gaming emulators are being run in one place, on RetroPie via EmulationStation due to its convenience and uniform solution. Most other cores still compile (and have binaries which are recent. (dosbox-staging is optional, so I suspect that is part of the reason for the incompatibility)

My primary question is whether RetroPie will transition to bullseye or bookworm, so that the gaming setup can be done in one place as opposed to being fragmented. I'd rather not upgrade the distro separetely, before it is officially supported, due to the large number of dependencies and platforms being supported...

Drag0nFly, 2025-06-28:
Thanks, appreciate that link and the supplementary information. It is something _I can_ live with until bookworm has been “certified“ to compile and run the majority of the cores, just thought it might be useful to check if there was a simpler fix. I was playing with the idea to simply upgrade in-place and test, switching SD-cards, and modifying the sources.list

I also thought about compiling a newer gcc on buster, but might not be worth the extra hassle (esp. considering that I am chasing some very granular DOSBox improvements), and would possibly introduce other incompatibilities (wouldn't be the only distro I'd replaced the compiler on though). I was surprised at how much better dosbox-staging ran though compared to the others (sdl2, dosbox-x, and “regular” dosbox)

In any case, it would of course be good to modernize it, and perhaps also have a 64 bit kernel as default (on the rpi4, in my case).

Thanks again for the input, I guess one basically has to wait it out (or go unsupported) in the mean-time to test this.