RetroPie/RetroPie-Setup #4086: **Title:** Splashscreen (VLC) Fails to Display Video on GPi Case 2 (CM4) – Video Output Forced to HDMI

MrGoatastic, 2025-07-07:
**Description:**  
After the `omxplayer` → `VLC` upgrade for splashscreens, the **video output no longer appears on the GPi Case 2’s LCD** (though audio now works). Testing suggests VLC defaults to HDMI output, ignoring the GPi Case’s custom display configuration.  

### **Steps to Reproduce** (GPi Case 2 + CM4, undocked):  
1. Fresh install RetroPie **4.8** via Raspberry Pi Imager.  
2. Run the **RetroFlag GPi Case 2 script** (switches between `config_hdmi.txt`/`config_lcd.txt` dynamically). 
    -  [ RetroFlag | GPiCase2-Script ](https://github.com/RetroFlag/GPiCase2-Script)
3. Boot RetroPie, configure Wi-Fi, install the **safe shutdown script** (restores audio). 
```bash
wget -O - "https://raw.githubusercontent.com/RetroFlag/GPiCase2-Script/main/retropie_install_gpi2.sh" | sudo bash
```
4. Enable a video splashscreen (e.g., `.mp4` with audio).     - **Observe (Pre-Upgrade):** Video plays on LCD, **no audio**.  5. Run `retropie_setup.sh` to upgrade packages (triggers VLC transition). Reboot.  
   - **Observe (Post-Upgrade):** **No video on LCD**, but audio works (likely via HDMI).  

### **Expected Behavior:**  
Splashscreen video *and* audio should play on the GPi Case 2’s LCD.  

---

### **Critical Context:**  
- **Hardware:** Retro Flag GPi Case 2 (**Raspberry Pi CM4**), used **undocked** (LCD only).  
- **Display Management:**  
  - The RetroFlag script dynamically swaps `config.txt` with either `config_hdmi.txt` (docked) or `config_lcd.txt` (undocked).  
  - **Issue Hypothesis:** VLC ignores the active `config_lcd.txt` and defaults to HDMI for video output, while audio routes correctly.  
- **Prior State:**  
  - `omxplayer`: Video worked on LCD (no audio).  
  - `VLC`: Audio works, but video fails (LCD unused).  

### **User Note:**  
*Not tested in docked mode. Issue likely stems from VLC’s output detection bypassing the GPi Case’s dynamic display switching.*

cmitu, 2025-07-07:
Please use the forum for support questions.

closed 2025-07-07 by cmitu

MrGoatastic, 2025-07-07:
This isn't a question. This is a problem on the way output is managed is scripts in the retropie_setup.sh for the splashscreen. You shouldn't have close that right away.

cmitu, 2025-07-07:
> Terminal output for the command `/opt/retropie/supplementary/kmsxx/kmsprint`

OK, so the system has 2 connected displays, one on the 1st HDMI port (HDMI-1) and one on the over DSI (DSI-1). Don't know how `vlc` picks the connected display to show the video and why if favours HDMI over DSI, but running with verbose logging may offer some info:
```
vlc -vv --intf dummy  --no-video-title-show --play-and-exit --mmal-layer 10001 /home/pi/RetroPie/splashscreens/test.mp4

```

Can you also post the `config.txt` that was used to add the DSI screen ?

MrGoatastic, 2025-07-07:
Here's a pastebin link. other whise i share the file also if you prefer.

[Pastebin for config_lcl.txt content](https://pastebin.pl/view/ba2dfdb0) wich remplace the current config.txt when its start as handheld mode if previous was the hdmi(dock) one. so basicly its the current config.txt as that what i'm trying to make work. i rarely plug it on dock.

[config_lcd.txt](https://github.com/user-attachments/files/21110624/config_lcd.txt)

Here for the verbose. it's kinda big too. i've shorten a part as it was repeat thousand of time the same line.

here's a [Pastebin of the log](https://pastebin.pl/view/bcfd3ef5) or use the summary bellow 🥲 

<details>
  <summary>
    Verbose of VLC ( really long )
  </summary>

```bash
pi@GPi-Retro:~ $ vlc -vv --intf dummy  --no-video-title-show --play-and-exit --mmal-layer 10001 /home/pi/RetroPie/splashscreens/test.mp4
VLC media player 3.0.17.4 Vetinari (revision 3.0.13-8-g41878ff4f2)
[00ba3b58] main libvlc debug: VLC media player - 3.0.17.4 Vetinari
[00ba3b58] main libvlc debug: Copyright © 1996-2022 the VideoLAN team
[00ba3b58] main libvlc debug: revision 3.0.13-8-g41878ff4f2
[00ba3b58] main libvlc debug: configured with ./configure  '--build=arm-linux-gnueabihf' '--prefix=/usr' '--includedir=${prefix}/include' '--mandir=${prefix}/share/man' '--infodir=${prefix}/share/info' '--sysconfdir=/etc' '--localstatedir=/var' '--disable-option-checking' '--disable-silent-rules' '--libdir=${prefix}/lib/arm-linux-gnueabihf' '--runstatedir=/run' '--disable-maintainer-mode' '--disable-dependency-tracking' '--disable-debug' '--config-cache' '--disable-update-check' '--enable-fast-install' '--docdir=/usr/share/doc/vlc' '--with-binary-version=3.0.17.4-0+deb10u2+rpt1' '--enable-a52' '--enable-aa' '--enable-aribsub' '--enable-avahi' '--enable-bluray' '--enable-caca' '--enable-chromaprint' '--enable-chromecast' '--enable-dbus' '--enable-dca' '--enable-dvbpsi' '--enable-dvdnav' '--enable-faad' '--enable-flac' '--enable-fluidsynth' '--enable-freetype' '--enable-fribidi' '--enable-gles2' '--enable-gnutls' '--enable-harfbuzz' '--enable-jack' '--enable-kate' '--enable-libass' '--enable-libmpeg2' '--enable-libxml2' '--enable-lirc' '--enable-live555' '--enable-mad' '--enable-matroska' '--enable-mod' '--enable-mpc' '--enable-mpg123' '--enable-mtp' '--enable-ncurses' '--enable-notify' '--enable-ogg' '--enable-opus' '--enable-pulse' '--enable-qt' '--enable-realrtsp' '--enable-samplerate' '--enable-sdl-image' '--enable-sftp' '--enable-shine' '--enable-shout' '--enable-skins2' '--enable-sndio' '--enable-soxr' '--enable-spatialaudio' '--enable-speex' '--enable-svg' '--enable-svgdec' '--enable-taglib' '--enable-theora' '--enable-twolame' '--enable-upnp' '--enable-vnc' '--enable-vorbis' '--enable-x264' '--enable-x265' '--enable-zvbi' '--with-kde-solid=/usr/share/solid/actions/' '--disable-d3d11va' '--disable-decklink' '--disable-directx' '--disable-dsm' '--disable-dxva2' '--disable-fdkaac' '--disable-fluidlite' '--disable-freerdp' '--disable-goom' '--disable-gst-decode' '--disable-libtar' '--disable-macosx' '--disable-macosx-avfoundation' '--disable-macosx-qtkit' '--disable-microdns' '--disable-mfx' '--disable-opencv' '--disable-projectm' '--disable-schroedinger' '--disable-sparkle' '--disable-srt' '--disable-telx' '--disable-vpx' '--disable-vsxu' '--disable-wasapi' '--enable-alsa' '--enable-dc1394' '--enable-dv1394' '--enable-libplacebo' '--enable-linsys' '--enable-nfs' '--enable-omxil' '--enable-udev' '--enable-v4l2' '--enable-wayland' '--enable-vcd' '--enable-smbclient' '--disable-oss' '--disable-crystalhd' '--disable-mmx' '--disable-sse' '--enable-neon' '--enable-mmal' '--enable-mmal-avcodec' '--disable-vdpau' '--disable-libva' '--enable-dav1d' '--disable-aom' '--disable-altivec' 'build_alias=arm-linux-gnueabihf' 'CFLAGS=-g -O2 -fdebug-prefix-map=/build/vlc-iTPUTR/vlc-3.0.17.4=. -fstack-protector-strong -Wformat -Werror=format-security ' 'LDFLAGS=-Wl,-z,relro -Wl,-z,now -Wl,--as-needed' 'CPPFLAGS=-Wdate-time -D_FORTIFY_SOURCE=2' 'CXXFLAGS=-g -O2 -fdebug-prefix-map=/build/vlc-iTPUTR/vlc-3.0.17.4=. -fstack-protector-strong -Wformat -Werror=format-security ' 'OBJCFLAGS=-g -O2 -fdebug-prefix-map=/build/vlc-iTPUTR/vlc-3.0.17.4=. -fstack-protector-strong -Wformat -Werror=format-security'
[00ba3b58] main libvlc debug: searching plug-in modules
[00ba3b58] main libvlc debug: loading plugins cache file /usr/lib/arm-linux-gnueabihf/vlc/plugins/plugins.dat
[00ba3b58] main libvlc debug: recursively browsing `/usr/lib/arm-linux-gnueabihf/vlc/plugins'
[00ba3b58] main libvlc debug: plug-ins loaded: 501 modules
[00ba3b58] main libvlc debug: opening config file (/home/pi/.config/vlc/vlcrc)
[00ba3d78] main logger debug: looking for logger module matching "any": 4 candidates
[00ba3d78] main logger debug: using logger module "console"
[00ba3b58] main libvlc debug: translation test: code is "C"
[00c09a68] main keystore debug: looking for keystore module matching "memory": 4 candidates
[00c09a68] main keystore debug: using keystore module "memory"
[00ba3b58] main libvlc debug: CPU has capabilities ARM_NEON FPU
[00bff4c8] main input debug: Creating an input for 'Media Library'
[00bff4c8] main input debug: Input is a meta file: disabling unneeded options
[00bff4c8] main input debug: using timeshift granularity of 50 MiB
[00bff4c8] main input debug: using default timeshift path
[... cut ...]
[00c2ec78] main playlist export debug: looking for playlist export module matching "export-xspf": 4 candidates
[00c2ec78] main playlist export debug: using playlist export module "export"
[00c2ec78] main playlist export debug: removing module "export"
[00bfdbf0] main playlist debug: deleting item `Media Library'
[00bfdbf0] main playlist debug: deleting item `test.mp4'
[00bfdbf0] main playlist debug: deleting item `Playlist'
[00c09a68] main keystore debug: removing module "memory"
pi@GPi-Retro:~ $ 
```
</details>

cmitu, 2025-07-08:
> [config_lcd.txt](https://github.com/user-attachments/files/21110624/config_lcd.txt)
> 

 Ok, I'll see if I can reproduce the display configuration and check if `vlc` can be configured to use the DSI video as primary output. Unfortunately the `vc4-fkms-v3d` overlay has no option to disable the HDMI port (as opposed to `vc4-kms-v3d`, which has a `nohdmi` parameter that disables the port).

> Here for the verbose. it's kinda big too. i've shorten a part as it was repeat thousand of time the same line.

Unfortunately the log is not useful, since you've connected an X11 display through your MobaXterm connection (you probably have X11 forwarding enabled ?) and `vlc` has chosen the `x11` output - that's not the same configuration that's present during boot:
```
....
[ae659ba0] main generic debug: looking for vout display module matching "opengles2": 15 candidates
[ae659448] main vout display debug: VoutDisplayEvent 'resize' 1920x1080
[ae65b4e0] main gl debug: looking for opengl es2 module matching "any": 2 candidates
[ae65b4e0] egl_x11 gl debug: EGL version 1.4 by Mesa Project
[ae65b4e0] egl_x11 gl debug:  extensions: EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_context_flush_control EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image_base EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_query_driver
[ae65b4e0] egl_x11 gl debug: No dma_buf_import - fall back to X
[ae65b4e0] main gl debug: no opengl es2 modules matched
[ae659ba0] main generic debug: no vout display modules matched
[ae659448] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[ae659ba0] main generic debug: looking for vout display module matching "xcb_x11": 15 candidates
[ae659448] main vout display debug: VoutDisplayEvent 'resize' 1920x1080
[ae659ba0] xcb generic debug: connected to X11.0 server
[ae659ba0] xcb generic debug:  vendor : Moba/X
[ae659ba0] xcb generic debug:  version: 12101015
[ae659ba0] xcb generic debug: using screen 0xf3
[ae659ba0] xcb_x11 generic debug: using X11 visual ID 0x21
[ae659ba0] xcb_x11 generic debug:  24 bits depth
[ae659ba0] xcb_x11 generic debug:  32 bits per pixel
[ae659ba0] xcb_x11 generic debug:  32 bits line pad
[ae659ba0] xcb_x11 generic debug: using X11 window 00280000
[ae659ba0] xcb_x11 generic debug: using X11 graphic context 00280001
[ae659ba0] xcb_x11 generic error: shared memory (MIT-SHM) not available
[ae659ba0] xcb_x11 generic warning: display will be slow
[ae659ba0] main generic debug: using vout display module "xcb_x11"
....
````

You may want to unset `DISPLAY` before running `vlc` so there's no X11 path in the way.

cmitu, 2025-07-08:
Ok, you don't need to re-run the `vlc` playback test with `DISPLAY` unset, I think it won't work anyway.

I've taken a look at the configuration (the LCD enabling `config.txt`) and at what arguments the splashscreen uses for playback. Here's what I found:

* If a HDMI display is connected (determined by runnning `tvservice -l`), the `vlc` will be run with the `--mmal-display hdmi-1/2` to display the video on the connected port. I think this is - mainly - the reason for which the DSI-1 connected display is not used by `vlc` for playback.
* 
* I used your configuration on a Pi4 (don't have CM4 nor a GPICase), but with that configuration I get no HDMI port enabled. Neither `kmsprint`, not `tvservice -l` show a HDMI port connected
````
$ /opt/retropie/supplementary/kmsxx/kmsprint
 Connector 0 (89) DSI-1 (connected)
  Encoder 0 (88) DSI
    Crtc 0 (87) 640x480@59.99 24.000 640/41/40/41/+ 480/18/9/18/+ 60 (59.99) P|D
      Plane 0 (31) fb-id: 95 (crtcs: 0) 0,0 640x480 -> 0,0 640x480 (XR24 AR24 RG16 RG24 BG24 YU16 YU12 YV12 NV12 NV21 P030)
        FB 95 640x480

$ tvservice  -l
1 attached device(s), display ID's are :
Display Number 0, type Main LCD
````
 I can't really test if `vlc` outputs something, since I don't really have a display over DSI connected, but `vlc` errors out anyway because of the audio configuration which references HDMI.

* `vlc` has the `--mmal-display` argument to choose the display, but the only documented (and probably accepted) values are HDMI-1/HDMI-2. According to https://github.com/RPi-Distro/vlc/issues/37#issuecomment-1198550991, if only the DSI-1 output would be connected, then the video playback would work, though I can't confirm that's true.

Given the above, I think that if you manage to produce a `config.txt` which will result in only the `DSI-1` output to be enabled & connected (i.e. no HDMI active), then the playback might work again. Otherwise, given the fact that the MMAL video output is deprecated and not recommended in recent RaspiOS releases, your best chance would be to use `omxplayer` again and create your own splashscreen service to display the video.

MrGoatastic, 2025-07-10:
Apologies for the delayed response—life gets busy sometimes! In the meantime, I’ve had a chance to test a few things.

I can confirm that both audio and video over HDMI are working correctly when the device is booted while connected to a TV via the dock. As suspected (and as you confirmed), it seems VLC will not output to DSI unless we can explicitly force it. Sadly, i cannot disconnect it as its part of the case connection with the CM4 module.

I’ll explore the possibility of disabling HDMI entirely using the LCD script, as you suggested. I also have attempted to revert to OMX, but ran into a separate issue—audio didn’t work, although that might be an easier fix. We’ll see!

cmitu, 2025-07-10:
> ...
> Sadly, i cannot disconnect it as its part of the case connection with the CM4 module.

I meant disabling the HDMI port via configuration, not by physically removing or obstructing it.

> i've tested a playback with omxplayer
> ...
> Result : video and sound. but rolling back to omx mean i can't update my retropie anymore. 🥲 at least its a safe fallback.

As I mentioned before, you can create your own splashscreen service/script to run, using `omxplayer`, and disable/remove RetroPie's splashcreen. This way you still get the splash working on DSI (not only on HDMI) and can still update.

1. Create a file in `/etc/systemd/system/omxplayer.service` (needs `root`) and add to it:    
    ```
    Description=Show custom splashscreen
    DefaultDependencies=no
    After=dbus.service
    ConditionPathExists=/home/pi/RetroPie/splash-omxplayer.sh

    [Service]
    Type=oneshot
    User=pi
    ExecStart=/home/pi/RetroPie/splash-omxplayer.sh
    RemainAfterExit=yes

    [Install]
    WantedBy=sysinit.target
    ```
   then run `sudo systemctl daemon-reload && sudo systemctl enable omxplayer.service` to make it available. 

2. Create the splashscreen script in `/home/pi/RetroPie/splash-omxplayer.sh` (no `root` needed here) from where you can call `omxplayer ....` to play your splashscreen.

NB: `omxplayer` will only work with Buster - current RetroPie 4.8 image - and it's no longer available starting with the RaspiOS Bullseye.

MrGoatastic, 2025-07-12:
> I meant disabling the HDMI port via configuration, not by physically removing or obstructing it.

Yes, I've tried, but seems the system ignore it and still send it on HDMI.

> You may try to replace dtoverlay=vc4-fkms-v3d with dtoverlay=vc4-kms-v3d,nohdmi for the LCD configuration screen and see if it works

Sadly, it don't work. I've already tried and i lose all video output.

Thanks you for the support, i really though we could get something out of it and stay on vlc ! I guess we can close the issues as it's "unfixable" in the actual way we should fix it. Potential fixes have all big drawback, like stuck on Buster, etc.