RetroPie/RetroPie-Setup #4087: Building RetroPi on Odroid has some errors.

Draqun, 2025-08-03:
Hi. I want to build a RetroPi for my odroid c1, however during the build not everything goes according to plan. Below I paste the log of the file ~/RetroPie-Setup/tmp/build/kmsxx/build/meson-logs/meson-log.txt
```bash

Build started at 2025-08-03T11:23:47.265963
Main binary: /usr/bin/python3
Python system: Linux
The Meson build system
Version: 0.45.1
Source dir: /home/odroid/RetroPie-Setup/tmp/build/kmsxx
Build dir: /home/odroid/RetroPie-Setup/tmp/build/kmsxx/build
Build type: native build

meson.build:45:7: ERROR: lexer
summary({
                 ^
```

Does anyone know how to fix it?
BTW. I follow your instructions from https://retropie.org.uk/docs/Odroid-C1-C2/

cmitu, 2025-08-04:
> BTW. I follow your instructions from https://retropie.org.uk/docs/Odroid-C1-C2/

They may be out-of-date, especially with regards to the OS image needed. What version of Ubuntu/Debian are you using and which image ?

Draqun, 2025-08-04:
As it happens, Armbian does not support HDMI on the Odroid C1, so that's out. I managed to install Ubuntu 20.04, but there were problems with dependencies. I am currently trying to match one of the software versions to 18.04. However, if I had been given clear instructions on which version would definitely work, it would have saved me a lot of trouble.